neroroxxx/BMC #3: LCD/TFT documentation

alexanderlarin, 2022-05-03:
I went through the entire docs files and couldn't find any references about LCD connection or customization (or it seems to be so).
Is there a way to understand this feature for further usage or something like a description of the default behavior?

neroroxxx, 2022-05-03:
Hi, BMC doesn't handle anything to do with Displays, it does have a class available to deal with OLEDs but it's just a basic driver, BMC itself doesn't render anything to the displays (mainly due to the fact that there are so many options out there for hardware and libraries and BMC being so scalable), you can get data to render on the display using the API, i know the documentation is not perfect as there is so much to cover that i've had a hard time keeping up with it but little by little i'm updating it, what exactly are you looking to print to a display? once i know i can put in some time today and maybe add a new doc for people using displays to give direct references to how to retrieve data from BMC for displays/

alexanderlarin, 2022-05-17:
@neroroxxx thank you for answering the questions. I spent a lot of time surfing the best way to configure my own midi project from scratch. This project looks great, suppose I will start from this point and it would be great if I have a strong connection with you and may ask some questions during the project. Also, I would be glad to help you in developemnt if you need additional resources (for example make configuration app open source and improve its UI or from the side of controller programming).