neroroxxx/BMC #4: Can not compile sample project on Win or Linux

Asynchronous003, 2022-11-27:
tested on win 11 home, and Endeavor OS(Arch Linux)
using Arduino 1.8.19, 2.0.0, 2.0.2 (BMC instructions followed)
board used: Teensy 4.1
error:
```In file included from c:\Users\djasy\OneDrive\Documents\Arduino\libraries\BMC\src/BMC.h:45:0,
                 from C:\Users\djasy\AppData\Local\Temp\.arduinoIDE-unsaved20221027-14856-1r7nia4.y67h\first-bmc-sketch\first-bmc-sketch.ino:42:
c:\Users\djasy\OneDrive\Documents\Arduino\libraries\BMC\src/utility/BMC-Def.h:137:6: error: #error "BMC is only Compatible with Teensy 3.2, 3.5, 3.6, 4.0, 4.1"
     #error "BMC is only Compatible with Teensy 3.2, 3.5, 3.6, 4.0, 4.1"
      ^
c:\Users\djasy\OneDrive\Documents\Arduino\libraries\BMC\src/utility/BMC-Def.h:145:4: error: #error "BMC is only Compatible with Teensy 3.2, 3.5, 3.6, 4.0, 4.1"
   #error "BMC is only Compatible with Teensy 3.2, 3.5, 3.6, 4.0, 4.1"
    ^

exit status 1

Compilation error: exit status 1```

neroroxxx, 2022-11-27:
Im not by my computer right now but im pretty sure that message is there if the selected board isn't any of the boards that bmc is compatible with, can you double check that u do have the 4.1 selected in the arduino ide?

Asynchronous003, 2022-11-27:
'Teensy 4.1' (BMC) is selected, and loading stock teensy examples works, Sorry, I should have added more info. looking at BMC-Def.h, the include arduino.h can't be found, when I downloaded ArduinoCore-avr-master and point to arduino.h, then vs-code error reads ```cannot open source file "stdlib.h" (dependency of "C:\Users\djasy\Downloads\ArduinoCore-avr-master\ArduinoCore-avr-master\cores\arduino\Arduino.h")C/C++(1696)``` 
I only know enough to get into trouble, the instructions were very clear, maybe i am missing something obvious, will be Awesome whem it works :)

neroroxxx, 2022-11-27:
Mmmm is teensyduino installed? I did a quick check on the github and the error points to the teensyduino not being installed, u have to install teensyduino AND the bmc for teensy boards which u do seem to have installed

Asynchronous003, 2022-11-27:
these .jsons are in additional  board manager URLS, and downloaded:
```https://raw.githubusercontent.com/neroroxxx/package_badassmidi_index/master/package_badassmidi_index.json
https://www.pjrc.com/teensy/package_teensy_index.json```

Asynchronous003, 2022-11-27:
Teensy by Paul Stoffregen Version 1.57.1
BMC Version 1.0.0

neroroxxx, 2022-11-27:
Ok i will have to wait to test it on my computer, which instructions did u follow to instal everything?

neroroxxx, 2022-11-27:
Ok another thing i noticed, are you using vscode to compile or the arduino ide?

Asynchronous003, 2022-11-27:
sorry, no, i just used it to poke around on windows, I'm using the Arduino-ide. Its so strange I get the same error on another computer.

Asynchronous003, 2022-11-27:
https://github.com/neroroxxx/BMC/blob/master/docs/installation-instructions.md
https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/
To install Teensy on Arduino IDE 2.0.0, click File > Preferences.  In “Additional boards manager URLs”, copy this link:
https://www.pjrc.com/teensy/package_teensy_index.json
In the boards manager column, type “teensy” in the “Filter your search…” box.  When Teensy appears, hover your mouse to expand and then click the INSTALL button.
Step 1: Install the BMC for Teensy boards
The BMC for Teensy boards are identical to the boards from teensy, they are a reference to the Teensy installation and require you to have teensyduino installed. These boards can only be used with BMC since they will require your sketches to have a config.h file in the sketch's folder, otherwise they will not compile.
On Arduino open Preferences, at the bottom of the preferences window you will see Additional Boards Manager URLs next to it a text field and next to the text field a button, click that button to open a list of your board manager's urls, there add a new line and copy and paste the url below:
https://raw.githubusercontent.com/neroroxxx/package_badassmidi_index/master/package_badassmidi_index.json
Next go to the Arduino Tools menu and select Board > Boards Manager, in the search field type "BMC for Teensy" and click install.

neroroxxx, 2022-11-27:
ok i got it, looks like its only with teensy 4.1 boards, the quick fix would be to go to documents/arduino/libraries/BMC/src/utility/BMC-Def.h and on line 50 replace "#el#if" with "#elif" idk how oh earth i did that... ill add an updated fix now as well, thank you for the heads up on this

Asynchronous003, 2022-11-27:
Awesome!!! thank you for all your work, I'm really Stoked I found BMC.
peace.

neroroxxx, 2022-11-27:
If you can just confirm that you are able to compile and i will close the issue

Asynchronous003, 2022-11-27:
All Good :)

closed 2022-11-27 by neroroxxx