neroroxxx/BMC #7: Keypad Option in Config File Maker not working properly

mateomodar, 2024-03-04:
The Keypad Option in the Config File Maker not working properly. Every time i insert a number of rows and columns over 4 or sometimes 5 (its inconsistent) it says too many pins used in the hardware options afterwards. When i get back it says for example 5 rows x 4 columns instead of the before selected 8 rows x 5 columns. Thanks for any help in advance, i know this is a new feature!

neroroxxx, 2024-03-05:
Hi, can you get a screenshot of the error message? i see the bug that shows the wrong number of keys which i can fix.

mateomodar, 2024-03-05:
[image]
This is how i configerated the rows and columns
[image]
This is how the error message then looks, 40 keys are selected. In global there is just one button and one encouder, shouldnt be too much pins.  The 16 pot inputs are with a multiplexer. 
[image]
When i go back to previous page it says in this case 8 rows x 4 columns, even tho 8x5 where selected before,

neroroxxx, 2024-03-05:
Thank you for the info, all fixed on editor version 2.2.1

neroroxxx, 2024-03-05:
Thank you for the info, all fixed on editor version 2.2.1, should automatically update on the browser just make you are running version 2.2.1

closed 2024-03-05 by neroroxxx