neroroxxx/BMC #8: Teensy 4.1 config.h wont compile

mateomodar, 2024-03-05:
After finishing the config process for teensy 4.1 nothing happens when pressing safe and continue. with the same options selected this works for other boards for example teensy 4.0

neroroxxx, 2024-03-05:
if you are referring to the "Save Changes & Continue" those are on all pages of the config file maker, which page specifically is stuck on? I just tested it and it all works, if you are stuck on the very last page where at the top left the title is OUTPUT then can you please copy and paste the text on the right box (white background) and paste it here.

To help debug you could also just right click on the browser and select "Inspect" then on the window that opens click the tab that says "Console" and take a screenshot of any error messages, that would help a ton.

mateomodar, 2024-03-06:
Tried it multiple times before, now it works like a charm! Thanks for that great software and the help!

closed 2024-03-06 by mateomodar