neroroxxx/BMC #9: Arduino IDE 2.3.2 wont compile

mateomodar, 2024-03-06:
When trying to compile first_bmc_sketch with my own config.h it gives me a compilation error
I use:
Arduino IDE 2.3.2
Teensyduino Boards: 1.59.0
Latest BMC Library: 2.2.2
BMC Boards: 1.58.1
wont work for Teensy 4.1 also tried for Teensy 4.0
I also tried to downgrade Teensyduino to 1.58, 1.57, 1.56, i allways uninstalled everything and then reinstalled with the right order like in the installation guide. That gave different kinds of errors. Also tried to downgrade BMC library with same results. 
The boards.txt MOD is not up to date anymore, right? the config.h file is saved in the right directory and reopens correctly when reopening the sketch.

error file: 
[compilation_error.txt](https://github.com/neroroxxx/BMC/files/14510923/compilation_error.txt)

neroroxxx, 2024-03-06:
it should work if you revert to teensyduino 1.58.2 for now, i'll have a patch to the BMC for Teensy Boards today.

The version of BMC for Teensy boards package has to always be the same.

mateomodar, 2024-03-06:
For now i get this error if i try to compile with teensyduino 1.58.2 
deinstalled everything and reinstalled in right order as in the installation guide.

[compilation_error_1.58.2.txt](https://github.com/neroroxxx/BMC/files/14511328/compilation_error_1.58.2.txt)

neroroxxx, 2024-03-06:
can you attach your config.h file?

mateomodar, 2024-03-06:
Sure 

[config_h.txt](https://github.com/neroroxxx/BMC/files/14511869/config_h.txt)

neroroxxx, 2024-03-06:
OK, so to fix that issue temporarily you have to disable DEBUG in the config file maker.

The compilation error is now fixed in version 2.2.3 of the BMC Library.

The issue with Teensyduino 1.59.0 is also fixed you can update the BMC for Teensy boards to version 1.59.0 if to use Teensyduino 1.59.0.

The Arduino library manager and boards manager doesn't update right away so before you update to Teensyduino 1.59.0 make sure that BMC for Teensy boards 1.59.0 is already showing as a version that's available.

closed 2024-03-06 by neroroxxx