neroroxxx/BMC #10: Bug in BMC-MuxInAnalog74HC40XX.h and problems with connecting to programmed Teensy 4.1

mateomodar, 2024-03-08:
When i use 74hc4051 i get this error:
[error_message.txt](https://github.com/neroroxxx/BMC/files/14539933/error_message.txt)
Also when i dont use a 74hc4051 the upload works just fine but BMC 2.2.3 seems to do nothing when i try to connect to the programmed Teensy4.1. When i try to load the button led example project i can connect just fine.

neroroxxx, 2024-04-04:
can you attach the config.h file with the 74HC40? sorry I completely missed this one

mateomodar, 2024-04-21:
Hey, also wasnt on the PC for some days, here is the config.h file:
[configh.txt](https://github.com/neroroxxx/BMC/files/15054608/configh.txt)

neroroxxx, 2024-04-28:
Thank you for reporting the bug, this is fixed in version 2.2.4

closed 2024-04-28 by neroroxxx