neroroxxx/BMC #12: Mini Display CS MCP23017

deepMago, 2024-08-18:
Hello,
althought I able to compile with attached file, I'm not able after to connect to Editor.
I Think after several modification to config.txt it's because you cannot use external chip (in my case MCP23017) to drive mini Display CS pins.
If my assumption is correct, do you think it's possible in future releases to add that option?
[configNotWorking.txt](https://github.com/user-attachments/files/16650066/configNotWorking.txt)

neroroxxx, 2024-08-18:
I didn't realize that the config file maker lets you use external pins for the cs pin of the displays, if so then that's a bug, I never tried using muxes for display cs pins so I'm not sure if that's going to be a good idea, something I will have to test out first

neroroxxx, 2024-08-18:
As far as connecting to the editor can you look at the serial monitor after boot? That will tell you if BMC stopped running and why

deepMago, 2024-08-18:
First time using the serial monitor, this is what i get
[image]

neroroxxx, 2024-08-18:
Try unplugging from usb and then plugging back in with the serial monitor open, you should see a lot more info as long as you enabled debugging when compiling

deepMago, 2024-08-18:
Ok I reupload the sketch and noww that's what i see
[image]

neroroxxx, 2024-08-18:
Ok I think the issue is with the Arduino serial monitor, it can bug out, try using tytools https://github.com/Koromix/tytools/releases/tag/v0.9.9
Use the tyCommander app it's a serial monitor for teensy and works WAY better than the Arduino serial monitor, looks like something is def crashing and I'm assuming has to do with the cs pin of the displays

deepMago, 2024-08-18:
it stucks
[image]

neroroxxx, 2024-08-18:
Ok yes it's def the cs pin, the library that handles the displays uses digitalWrite and mix pins on BMC start at like pin 64 or so if I remember correctly so the library is trying to initialize a pin that doesn't exist

deepMago, 2024-08-18:
Ok, do you think I can try modify the code for mapping the extra pins or it's better to use the teensy pins? In the first case I could try in 24 hours cause in italy it's time to sleep.
Thank you

neroroxxx, 2024-08-18:
You'd have to remove all code that handles the cs pin in the display library, then you'd have to add code to manipulate the displays cs pins thruout BMC, it would be a big job to do that. That's one of the reasons it's supposed to be done using built in pins

neroroxxx, 2024-08-18:
I def recommend using teensy pins for displays no mux pins at all

deepMago, 2024-08-18:
Ok so Way better use the build in pins, thanks

closed 2024-08-21 by deepMago

neroroxxx, 2024-08-21:
let me know how its working for you and if you able to share some images of your build