neroroxxx/BMC #13: miniDisplay CC change color

deepMago, 2025-01-29:
Hi,
I cannot figure out how to do (or if it's not possible) a let's call "Stompbox mode".

Let's say we have a BMC with 3 miniDisplay and 3 buttons, no leds or pixels

We have a multiFX connected via MIDI with 3 FX slot (assume OD, DELAY, REVERB) that you could activated sending a MIDI CC (0 for Off, 127 ON), for FX1 CC00, for FX2 CC01, for FX3 CC02

Button1 send MIDI CC00 switching between 0 and 127, button2 send MIDI CC01 and Button3 send MIDI CC02

Every display shows the FX NAME (maybe using the Event name from library), so Display1 show "OD", Display2 show "DELAY" and Display3 show "REVERB"

Now I would like Displays act like a LED, so when CC is 127, I could invert background and text color or highlight the display to know the relative effect is on

neroroxxx, 2025-01-31:
i'll have to see if there's an option to streamline this already coded in.

neroroxxx, 2025-01-31:
Version 2.2.5 is live now and will give you the option. If you set the "Display Names" option on the display it will show the name of the event assigned to the display, You can create a MIDI Control Change event and assign it to the footswitch and the display. When the CC value is more than 64 it will invert, you could assign the background color by default to black and text to red so when the CC is 64 the background would be red and the text white.

Below are the steps, let me know if this works for you.

1. Create a "MIDI: Control Change" event, with Value at 0 and Value B at 127 (this will toggle when assigned to a button)
2. Assign a name to the event from the name library.
3. Assign the event to a button so it toggles the FX Processor's parameter.
4. Add the event to a display, which will show Channel/CC/Value by default.
5. Turn on the "Display Names" option for that display.
6. The display will now show the name you assigned to the event.
7. When you press the footswitch to toggle the CC value, the display will change color:
- If the CC value is 64 or more, the display inverts color.
- If the CC value is less than 64, it will show the assigned colors.

deepMago, 2025-02-01:
WOW!
It's works exactly as desidered, I would like to thank you for this great work. Now it's time to go look into the code to learn something new; in this two days I've tried to create my own modified code, but I wasn't able to do the right changes.
Hopefully have an official version it's the best!

deepMago, 2025-02-01:
[image]

closed 2025-02-01 by deepMago