neroroxxx/BMC #14: Editor 2.2.5 Drag and Drop not working

deepMago, 2025-02-04:
With the latest release of the editor on Mac (with Chrome) - not tested on widows - it's impossible to drag and drop events; instead on the empty slots appears several yellow blocks, others functions seems to work.

[image]

neroroxxx, 2025-02-04:
Thank you for reporting, there are many bugs that got cached from an old file, quit browser and reload the app, if you have it installed on chrome, uninstall it, quit browser and reinstall it, then please test it and confirm that it's working normally. Bug is fixed on my end so just gotta ensure that chrome uncaches the old files

deepMago, 2025-02-04:
Resolved now, what I do:

-cleaned Chrome  cache

-restarted the PC

closed 2025-02-04 by deepMago