neroroxxx/BMC #15: Teensy 4.1 mid sub host

deepMago, 2025-03-10:
I’m trying without success to connect an Hotone AMPERO Mini via USB  MIDI Host.
I’m sure the device receive midi, tested connected to a pc.
Also I Can confirm the host controller on Teensy 4.1 is active because I have 5 volts on VHST pin.
AMPERO mini have usb-c connector but I suppose it’s compatible with a USB-a; tried connect directly and with a self powered hub.
I don’t have other device to check if the port really communicates, tried to connect a HX stomp but I’m not sure is working with midi host.
I’m locked, any suggestions that could me help move in the right way?

neroroxxx, 2025-03-10:
after a quick search i see that the AMPERO doesn't specify that it supports MIDI over USB, only audio, in that case it won't work with the Teensy USB Host.

deepMago, 2025-03-11:
officially there no documentation about, but according to several posts online like this one:
https://www.thegearpage.net/board/index.php?threads/hotone-ampero-mini.2470102/page-5
All Ampero devices responds to USB MIDI as the Mini do if connected via USB -C to a PC using your MIDITool I'm able to send PC and CC command:

[image]

now I've to check the Serial+MIDI+Audio, also I have to find a device that is USB MIDI Compliant to check the USB Host. 
I've one more question, if I connect the USB Host to my Mac should I see a MIDI connection or because they are two Hosts nothing happen?

deepMago, 2025-03-11:
I'm happy to say now it's working. After update to 2.3.0 from 2.2.5 nothing happened.

So I've edit the configuration file in "Advanced" --> "Others" and changed "Use MIDIDevice _BigBuffer" to TRUE and now it works.

Good also to see a new version release!

closed 2025-03-11 by deepMago