neroroxxx/BMC #16: Problems with editor and SySex messages

AndreySom, 2025-04-29:
I'm experiencing problems with my editor on both Chrome browser and standalone app. My drop-down boxes are all in white with no characters, and it only appears when I roll my mouse over each option.
It is still possible to program it, but it slows me down a lot, since I can't see all the options at once.

https://github.com/user-attachments/assets/a5547d38-2bd9-43ea-8dfa-1ec828d5eb2f

My second issue is that I'm having a having a problem with custom SySex.

I'm controlling my zoom MG50-G using PC's and it works just fine.
It is possible to control individual stomps using Sysex, but in order to, I must send a Sysex starting with 7E (126) (F0 7E 00 56 00 F7) actually, but BMC already sends the F0/F7).
It is a 4 byte message,  but when I set the 1st byte to 7E and save it, it returns to 0 (0x00). I have already tried with version 2.2 and now with the latest version. 
I don't know if it is a problem or a limitation.

https://github.com/user-attachments/assets/c9cd958a-e281-44d7-a14b-e0c389ca84b3

Thanks for adding the ESP boards to BMC, I have a couple of them laying around and I'll give it a try.
Awesome work as always, thanks Nero

neroroxxx, 2025-04-29:
Thank you for the feedback, i will review it tomorrow and push fixes as soon as possible, the issue with the dropdowns is very strange but im assuming it has to do with the new styling. Can you confirm your version of chrome?

AndreySom, 2025-04-29:
I believe it´s related to the new visual style also. I have seen it in version 2.2, but my BMC was already running and programmed and it just didn't bother me.
I'm running it in windows 11

[image]

Thanks

neroroxxx, 2025-04-29:
ok i believe it's just windows and the whole light/dark modes, im updating those select boxes to be fully custom and that way they aren't os dependent, I have the fix already but i'll update the apps int he next day or so once i patch the issue with sysex

neroroxxx, 2025-04-30:
As far as the Custom SysEx issue goes, im not sure i understand the issue, i tested it and it's saving properly, so basically BMC will add the 0xF0 and 0xF7 automatically when sending the message, you only have to set the bytes in between, i've tested it on an older version of BMC and it works as expected to say the bytes,  can you confirm the editor version and library version you are running? 

the issue may be related to an older version of the library, i know there was an issue with certain values not properly saving but that should be fixed on the latest version of BMC

neroroxxx, 2025-05-07:
Version 2.3.4 is now available, the editor has new dropdowns that won't have that issue you were having, Custom SysEx don't seem to have that issue it may be an older version of the library that caused it.

closed 2025-05-07 by neroroxxx